BRL-CAD/brlcad #3: Generate manual pages from brlcad-docs asciidoc repo

brlcad, 2021-07-23:
The current Docbook XML files rely on XSLT and/or FOP, and are generated during compilation in the brlcad repo.  A new doc management system, held in a separate brlcad-docs repo, uses Asciidoc files and Antora for html generated output and Asciidoctor for manual pages.

A desirable end-state is having docs maintained in a separate brlcad-docs repo, in asciidoc format, but generated during compilation of the brlcad repository (e.g., as a git submodule).  This will require proper detection and/or supplying of Antora+Asciidoctor and any other tooling necessary to generate the desired manual page, html, and pdf outputs currently being generated by the Docbook XML system.  I likely necessity for this will be CMake detection logic to report necessary tools that must be preinstalled by developers, or providing required dependencies as managed dependencies. 

This issue is space for discussion on how to resolve migration issues.

dashohoxha, 2021-07-23:
If using Docker is an option, then it is possible to build a Docker container that includes all the tools necessary for generating the docs (Antora, asciidoctor, etc.) Or maybe there are already such containers out there.

However I am in favor of building and distributing docs as a separate package (as mentioned [here](https://github.com/BRL-CAD/brlcad/commit/bc5ff11154f0187a2cc180ff015d157066cc2fa0#commitcomment-53865991))

brlcad, 2021-07-23:
We can certainly use Docker, and we have in the past (at least I used it a number of years back), but we've never considered requiring it.

In general, we have an expectation that a user can check out a repo, install a few critical tools (like a compiler and cmake), detect system libraries (e.g., X11, FOP, etc), and we nearly provide the rest (e.g., libpng, tcl/tk, libz etc).  So in this perspective, we'll either need to 1) add to the list of critical tools, 2) not build docs when tools aren't available (including manual pages?), or 3) somehow provide a few more tools.

There are merits to each, but our historic perspective is to do #3, which as noted is complicated here a bit by the fact that asciidoctor is in ruby.

dashohoxha, 2021-12-09:
I think that there is no need to generate the man pages during the compilation, because they can be pre-generated and stored on the `brlcad-docs` repo. So, no additional tools or dependencies are needed.

starseeker, 2021-12-09:
I would still vastly prefer a solution that is integrated with our build.  A non-self-contained solution implicitly depends on a much more complex set of external tools, and is more prone to breakage or obsolescence - particularly in time scales measured in decades.  Something we can integrate with our build directly not only allows us to ensure the complete tool chain remains functional as long as we require it, and also helps to make documentation part of the "core" workflow.  If documentation is a separate system, it's just that much easier to ignore it.

Sean mentioned translating asciidoctor to C/C++ - it would be interesting to know what that would take (in particular, how much additional functionality beyond just the asciidoctor code itself would need to be translated to achieve a fully functional stand-alone package.)  If that is a viable option and we can figure out a way to do it I could see using the external asciidoc tools as an interim solution to let documentation updating proceed, but I would like to have a viable plan in place to eventually get us back to a viable self-contained solution.

dashohoxha, 2021-12-09:
I cannot comment on your preferences.

However, regarding translation of asciidoctor to C/C++, despite what sean might say or wish, I don't see any chances of it happening. I am not able to do it, and I am not aware of anyone else working (or even thinking) about it.

Just letting you know, so that you can make informed decisions.

starseeker, 2021-12-09:
The other option would be to bootstrap Ruby and use a local copy to drive asciidoctor - if it's simple enough to set up a self-contained environment, the new build system work might make that viable.  I'll do some experiments and see what it looks like.

brlcad, 2021-12-09:
I agree that we want to keep generating the docs as there are numerous validation and integration benefits.   That said, I think docs will eventually *need* to live in a submodule regardless (with the main module pulling it in when configured to build docs) just like our sample db's.  Unless we're okay increasing the main repo by a gig, there're already docs not yet added like the new Intro to BRL-CAD, a rendering tutorial, some reports, and overview/showcase presentations, and their assets alone will dominate.  They could be offloaded to LFS, but that's not much different than a submodule imo, just not as convenient scripting-wise and doesn't address dependency management.

Solution to me has been to turn brlcad-docs into a submodule, give it a CMake build, have it detect antora/asciidoctor/ruby/node.js/etc, and have a build step generate the docs.  Yes, that means we will need to pre-install things to make a release, and I think that'll be fine.  Windows is the only real thorn there.  Devs making releases on windows already have to install MSVC, CMake, NSIS/Wix, so add a couple more to the list.

If that's hard to swallow, then we probably should adopt a system like Conan.io or Buckaroo since that's what they specifically do.  The idea of manually bootstrapping ourselves is less and less appealing.

starseeker, 2021-12-09:
If we're going to go with a brlcad-docs submodule, then would the way forward be to set up a separate brlcad-docs github project and establish the new setup there, and eventually replaces the docs dir in the main repo with the submodule when its ready?

brlcad, 2021-12-09:
> The other option would be to bootstrap Ruby and use a local copy to drive asciidoctor - if it's simple enough to set up a self-contained environment, the new build system work might make that viable. I'll do some experiments and see what it looks like.

Just be aware that only solves the manpage part.  Antora is needed for the rest.  Installation is trivial with Node LTS, but that's a different toolchain from what asciidoctor needs.  I do think our binary releases should keep shipping all three, man+pdf+html outputs, so the brlcad-docs repo could just have a batch+shell script that actually does bootstrapping before cmake as an interim solution.

dashohoxha, 2021-12-09:
> I do think our binary releases should keep shipping all three, man+pdf+html outputs

My humble opinion is that we don't need to ship the docs with the binary, if they are available online.
Anyway, man and HTML docs can be generated easily, so are easy to ship with the binary.

With PDF the situation is a bit more complex. Usually the tool `asciidoctor-pdf` can be used to generate a PDF from an asciidoc document. It can be installed with `apt install ruby-asciidoctor-pdf` or with `gem install asciidoctor-pdf`.
Man pages are almost standalone asciidoc files, so it is easy to generate a PDF for them with with `asciidoctor-pdf`. For example:

```bash
asciidoctor-pdf docs/modules/man/pages/1/mged.adoc
mv docs/modules/man/pages/1/mged.pdf pdf/
```

However, the rest of the Antora docs usually use some extra syntax and directives, besides those of asciidoc. For example including images, partials, examples, etc. So, `asciidoctor-pdf` fails to generate a PDF for them.

This issue (generating PDF files from Antora docs) seems to be still a work in progress in the Antora community: https://gitlab.com/antora/antora/-/issues/349

brlcad, 2021-12-10:
Various Government agencies use BRL-CAD within secured environments without online access.  It's a primary use case that we must maintain and support.  So, while docs can be in html and hosted online, they have to also ship with BRL-CAD and function offline (i.e., at least readable).

I'd come across https://www.npmjs.com/package/@nex/antora-pdf earlier as an Antora plugin, and it looked promising..  PDF isn't as important as manpage and html output -- just nice to have for better offline/printing capability.

dashohoxha, 2021-12-10:
> Various Government agencies use BRL-CAD within secured environments without online access. It's a primary use case that we must maintain and support. So, while docs can be in html and hosted online, they have to also ship with BRL-CAD and function offline (i.e., at least readable).

I understand.

To summarize:
- For all the pages (man pages and other docs) it is easy to generate online and offline HTML pages.
- For man pages, it is easy to generate them in MAN format, and also in PDF format if needed.
- For the rest of the pages, converting them to PDF might be possible (maybe using some tricks). But I am not sure if this is really needed (given that they are available as offline HTML). And also it is not immediately clear what should be included in a single PDF file:
  + All the docs?
  + A PDF file for each HTML page? (I don't think so.)
  + A PDF file for each book and each article? What about the rest?
     In this case some pre-processing tricks might be needed, before the PDFs can be generated.

dashohoxha, 2021-12-10:
I have modified the script [build.sh](https://github.com/BRL-CAD/brlcad-docs/blob/main/build.sh) to generate an offline HTML site as well.

Regarding dependencies, do you think it is feasible for you to install [podman](https://podman.io/getting-started/installation)?
Podman is similar to Docker but less heavy (does not use a service).

If you can install Podman, then I believe that I can build a Podman container that includes all the needed dependencies for generating the docs (nodejs, ruby, asciidoctor, etc.)  You will just call `./build.sh offline /usr/share/doc/brlcad` and it should build the container with all the dependencies and generate the docs.

dashohoxha, 2021-12-12:
> I will try to check this plugin and see how it works.

This plugin is still experimental: https://gitlab.com/djencks/antora-pdf/-/tree/master/packages/antora-pdf

Besides this, it extends Antora by overriding the default generator (something like: `antora --generator @djencks/site-generator-default ...`). But we are already using another extension to override the default generator, namely the one that provides search functionality (we do it like this: `antora --generator antora-site-generator-lunr ...`).
With the current version of Antora, extensions of the generator are done by overriding the whole thing. So, it is not possible to use both of these extensions at the same time, unless I build a custom extension myself that provides the functionality of both of these extensions.

However in the new version of Antora (that is expected to come out soon), I believe that handling of extensions has been improved, such that they can add functionality incrementally, instead of overriding the whole thing.
So, I am postponing this task (using the antora-pdf plugin for generating PDFs) for later.

dashohoxha, 2022-02-08:
> - For all the pages (man pages and other docs) it is easy to generate online and offline HTML pages.
> - For man pages, it is easy to generate them in MAN format, and also in PDF format if needed.

By the way, if you try to print a page from the browser (I have tried with Chrome) you get a wonderful preview, which can also be saved as PDF.