BRL-CAD/brlcad #45: Can this not run on Mac M1s?

StathisKap, 2022-05-17:
I've got the 14 inch 2021 model, so I have an M1 pro, which ofcourse in arm architecture. Is there any way to run BRL-CAD on this?

starseeker, 2022-10-06:
In principle it should be compilable there, although I think it's only been tried a few times and probably not within the last few months.

brlcad, 2022-10-07:
@StathisKap yes, it should build just fine.  If you tried to compile and encountered an error, please open an issue with the details.  The pre-built binary builds will obviously not work given the different architecture, so you will need to compile from a git checkout.  Instructions are available at https://brlcad.org/wiki/Compiling

closed 2022-10-07 by brlcad