BRL-CAD/brlcad #55: Not compiling, Linux Ubuntu 20.04 LTS

cvaisnor, 2022-08-17:
Cloned the repo, checked for all updated dependencies. Set cmake parameters according to Step 5 here: [Compiling](https://brlcad.org/wiki/Compiling)

I'm on an Intel i9 ten core. I tried 'make' in default and with -j10 flag. 

Dependencies:
gcc 9.4.0
g++ 9.4.0
make 4.2.1
cmake 3.16.3

I am including a screenshot of the terminal where compiling fails, as well as the logs. 
I am not sure how to move forward, thank you!

[GDAL_BLD-build-err.log](https://github.com/BRL-CAD/brlcad/files/9365636/GDAL_BLD-build-err.log)
[image]
[image]

brlcad, 2022-08-18:
Thank you for the report.  Is there a chance you ran cmake+make and then ran cmake again with different options?  That error is saying that libgdal was told to compile against a system libz, which it would do if you ran cmake without any options.  If you added -DBRLCAD_BUNDLED_LIBS=ON, it shouldn't be trying to compile against a system libz and the symbols would be different (they get prefixed).

To confirm / rule out that issue, would you delete+recreate your build dir, run:  cmake .. -DBRLCAD_ENABLE_STRICT=NO -DBRLCAD_BUNDLED_LIBS=ON -DCMAKE_BUILD_TYPE=Release 2>&1 | tee cmake.log

Then run "make 2>&1 | tee make.log"

That will capture the entire output from cmake and make to a cmake.log and make.log file respectively, which you can then post here if it still fails.

cvaisnor, 2022-08-24:
That worked and everything compiled correctly! 
Thank you!

closed 2022-08-24 by cvaisnor