BRL-CAD/brlcad #58: Failed build - no such file or directory

gimmemabrewski, 2022-09-18:
Seems to build fine but during install I get a failure

```
.......
-- Set runtime path of "/tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/bin/pixcmp" to "/opt/brlcad/lib:$ORIGIN/../lib"
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/bin/benchmark
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/bldg391.pix
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/m35.pix
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/moss.pix
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/sphflake.pix
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/star.pix
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/world.pix
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/bldg391.log
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/m35.log
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/moss.log
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/sphflake.log
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/star.log
-- Installing: /tmp/makepkg/brlcad/pkg/brlcad/opt/brlcad/share/pix/world.log
cp: cannot stat '/tmp/makepkg/brlcad/src/../../doc/legal/bdl.txt': No such file or directory
==> ERROR: A failure occurred in package().
    Aborting...
```

starseeker, 2022-10-06:
Is this using the PKGBUILD script for Arch Linux?

brlcad, 2022-10-07:
Presuming this is arch linux, I just pushed an update to the PKGBUILD file that should fix this error.  If you want to test a fix locally, edit the PKGBUILD file, find the two lines at the end referring to bdl.txt and bsd.txt, and replace srcdir with startdir.

closed 2022-10-07 by brlcad