BRL-CAD/brlcad #73: Headless/Non-interactive?

anonimo82, 2023-04-07:
Hello everybody.
I can't answer this question by Googling myself, so I'm asking, to be safely sure to get the "right" answer.
Can I use BRL-CAD from the Shell? If so, can I use it non interactively?
I mean, can I issue a command and have a database output without using [internal shell commands]/[scripting]?

Thanks for any help.

Regards =)

brlcad, 2023-04-08:
Yes, headless non-interactive is fully supported.  The MGED program can be run in just about every possible configuration including in a classic “command” mode where you can issue commands or sets of commands  1)on the command line, 2) via piped input, 3) via stdin, 4) via script.

for example, “mged -c file.g tops” just runs the tops command and exits.  Similarly, “echo search . -type comb | mged -c file.g” runs the search command and exits.  Another note detailed example showing multiple methods is available on the project wiki at https://brlcad.org/wiki/SGI_Cube

closed 2023-04-08 by brlcad

anonimo82, 2023-04-08:
Hey, thanks so much, exactly what I was looking for!

Kudos!!!!!!