BRL-CAD/brlcad #76: Build fails on M1 Mac

fancsali, 2023-04-30:
Following the [instructions](https://brlcad.org/wiki/Compiling) and trying to build BRL-CAD on a MacBook Pro with a M1 Max chip, running OSX 13.1 Ventura, results in a segfault and a failed build:

```
make -j8
[...]
[ 95%] Generating ../share/db/ktank.g
/bin/sh: line 1: 81579 Segmentation fault: 11  ../bin/asc2g /Users/daniel.fancsali/Private/brlcad-7.34.0/db/die.asc /Users/daniel.fancsali/Private/brlcad-7.34.0/build/share/db/die.g
make[2]: *** [share/db/die.g] Error 139
make[2]: *** Deleting file `share/db/die.g'
make[1]: *** [db/CMakeFiles/die.g.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 95%] Built target cube.g
[ 95%] Built target demo.g
[ 95%] Built target galileo.g
[ 95%] Built target kman.g
[ 95%] Built target ktank.g
[ 95%] Built target goliath.g
[ 95%] Built target havoc.g
make: *** [all] Error 2
```

Rather oddly, if trying to rebuild, it'll fail at the same stage, but at a different `.g` file.

starseeker, 2023-09-09:
@fancsali Does this still happen if you try building with the latest main and -DENABLE_ALL=ON set for CMake?

Elastino2, 2025-04-25:
# Update 2025
## Geogram Build Fail
### PoissonRecon
```
/brlcad/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/geogram/third_party/PoissonRecon/SparseMatrix.inl:197:22
```
`SetZero()` is calling to non-existing members.
```
 error: no member named 'm_N' in 'SparseMatrix<T>'
```
My solution : Comment out `SetZero()`
```
geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/geogram/third_party/PoissonRecon/PlyVertexMini.h:71:104: error: no member named 'value' in 'PlyOrientedVertex<Real>'
```

```
PlyOrientedVertex operator - ( PlyOrientedVertex p ) const { return PlyOrientedVertex( point-p.point , normal-p.normal ); }
```
Change `p.value` to `p.point` in `-` operator in Line 71 and Line 100.
This allows build of `GEOGRAM`.

## Manifold Build Fail
```
/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/utilities/../third_party/thrust/thrust/type_traits/is_contiguous_iterator.h:142:3: error: use of undeclared identifier '_VSTD'
```
My temporary solution : Change `_VSTD' to `std`(without any thoughts), then it ends up with,
```
-- stderr output is:
In file included from /Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/manifold/src/constructors.cpp:17:
In file included from /Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/cross_section/include/cross_section.h:17:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/functional:552:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__functional/boyer_moore_searcher.h:27:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:570:63: error: too many arguments provided to function-like macro invocation
  570 |       _NOEXCEPT_(__noexcept_move_assign_container<_Allocator, __alloc_traits>::value);
      |                                                               ^
/Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/utilities/../third_party/thrust/dependencies/libcudacxx/include/cuda/std/detail/libcxx/include/__config:1243:11: note: macro '_NOEXCEPT_' defined here
 1243 | #  define _NOEXCEPT_(x) noexcept(x)
      |           ^
In file included from /Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/manifold/src/constructors.cpp:17:
In file included from /Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/cross_section/include/cross_section.h:17:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/functional:552:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__functional/boyer_moore_searcher.h:27:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:569:86: error: expected ';' at end of declaration list
  569 |   _LIBCPP_CONSTEXPR_SINCE_CXX20 _LIBCPP_HIDE_FROM_ABI vector& operator=(vector&& __x)
      |                                                                                      ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:1309:61: error: too many arguments provided to function-like macro invocation
 1309 |     _NOEXCEPT_(__noexcept_move_assign_container<_Allocator, __alloc_traits>::value) {
      |                                                             ^
/Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/utilities/../third_party/thrust/dependencies/libcudacxx/include/cuda/std/detail/libcxx/include/__config:1243:11: note: macro '_NOEXCEPT_' defined here
 1243 | #  define _NOEXCEPT_(x) noexcept(x)
      |           ^
In file included from /Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/manifold/src/constructors.cpp:17:
In file included from /Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/cross_section/include/cross_section.h:17:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/functional:552:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/__functional/boyer_moore_searcher.h:27:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:1309:5: error: expected function body after function declarator
 1309 |     _NOEXCEPT_(__noexcept_move_assign_container<_Allocator, __alloc_traits>::value) {
      |     ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/vector:2001:63: error: too many arguments provided to function-like macro invocation
 2001 |       _NOEXCEPT_(__noexcept_move_assign_container<_Allocator, __alloc_traits>::value);
      |                                                               ^
/Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/utilities/../third_party/thrust/dependencies/libcudacxx/include/cuda/std/detail/libcxx/include/__config:1243:11: note: macro '_NOEXCEPT_' defined here
 1243 | #  define _NOEXCEPT_(x) noexcept(x)
      |           ^
In file included from /Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/manifold/src/constructors.cpp:17:
In file included from /Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD/src/cross_section/include/cross_section.h:17:
[... cut ...]
make[5]: *** [src/manifold/CMakeFiles/manifold.dir/src/constructors.cpp.o] Error 1
make[4]: *** [src/manifold/CMakeFiles/manifold.dir/all] Error 2
make[3]: *** [all] Error 2

CMake Error at /Users/lee/src/brlcad/build/bext_build/manifold/MANIFOLD_BLD-prefix/src/MANIFOLD_BLD-stamp/MANIFOLD_BLD-build-Debug.cmake:47 (message):
  Stopping after outputting logs.
```
Build was performed with 
```
clang --version
Apple clang version 17.0.0 (clang-1700.0.13.3)
Target: arm64-apple-darwin24.4.0
```
I wil try to do it again with `gcc` and/or build manifold only.

starseeker, 2025-04-25:
Can you also try building the latest Manifold upstream in isolation?  (https://github.com/elalish/manifold)  I've got the newest version working in a branch, which will be merged after the next release, and that might fare better compiling on your platform.  If so, you can give the manifold3 branch in the brlcad repo a try.

starseeker, 2025-04-25:
I still need to upgrade Geogram to the latest, but could you give their latest a try as well?  (https://github.com/BrunoLevy/geogram)  If their latest doesn't build correctly, that'd be the place to file a bug report for the above error and hopefully get it fixed upstream.

Elastino2, 2025-04-25:
# Update 2025-2
## Needs gcc instead of clang
Switching to `gcc-14` resolved many things due to GNU C++ extensions.
Indeed, cmake was passing `-std=gnu++17' to `clang++` but these GNU extensions are not included in MacOS SDK.

## bext, strclear build fail
Apple made a mistake.
As of MacOS SDK15.4, Apple used both C syntax and C++ syntax in single line.
And, I'm stuck on every builds, now.
GNU filed a bug to Apple. 
Hence, I will wait unit it gets fixed. 😄

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=119590

Elastino2, 2025-04-28:
> I still need to upgrade Geogram to the latest, but could you give their latest a try as well? (https://github.com/BrunoLevy/geogram) If their latest doesn't build correctly, that'd be the place to file a bug report for the above error and hopefully get it fixed upstream.

geogram(c02b15f5bfd236e749a2ce2c8ee31b00f00fe5bd) can be built successfully. No issue with M4.
manifold(54baa62b4ad49a8a4b82ebe25785cce14d622d40) can be built successfully. No issue with M4.

Elastino2, 2025-05-20:
# Update 2025-3
* Switched again back to `clang`, Apple clang version 17.0.0 (clang-1700.0.13.3)
* Completed `bext` build after replacing `manifold` with the most recent version without the `thrust`.
* Built `BRLCAD` only up to 22%, struggling with `common.h` filename issue. (e.g. `BRLCAD/include/common.h`, `manifold/include/common.h` and etc.)
* Stuck on building `librt` due to the above issue.
* Removed unexpected `@par` from `bg/plane.h` and `bu/cv.h` to achieve `bext` build, but have not provided commit for it.
* Removed unrecognized `NORETURN` keyword from `bu/magic.h`, possibly coming from `thrust`.
* Changed `b_off_t` to `off_t` in `bu/vls.h`
* Had to comment out access to non-existing `d`,`offset`,`center` in `libbg/spsr/Octree.inl`
* Had to apply same fix for `PlyVertexMini.h` in `libbg/spsr`, `p.value` to `p.point`
* Same comment out solution with `SparseMatrix.inl` for `Resize()` and `Solve()`
* When building `libicv`, `ppm.c` refers to system `#include "ppm.h"`, results `pm_config.h` not found. Changed it to `#include <netpbm/ppm.h>` to indicate system header. (In my case, `/opt/homebrew/include/netpbm/`)
* I'm on BRLCAD : 4ed69d75ef0c4465d52e28309c1e57e5df43389d, BEXT : affb3b60f63c9b77852512ce5ed8bdb801346638

Elastino2, 2025-08-21:
# Finally Built with M4.
[image]

```
[100%] Built target regress_mater
Done.

BRL-CAD Release 7.42.1, Build 20250821

Elapsed compilation time: 26 seconds 
Elapsed time since configuration: 20 minutes 52 seconds 
---
Run 'make install' to begin installation into /usr/brlcad/rel-7.42.1
Run 'make benchmark' to run the BRL-CAD Benchmark Suite

[100%] Built target buildtimedelta
```

# Clang
`Apple clang version 17.0.0 (clang-1700.0.13.5)`

# CMake
`cmake version 4.1.0`

# Git Hash
`868ad46c77a3c0097b9bfe828f6e46e46ed57a5a`

# Caveats
1. Had to add, `#include <malloc/_malloc_type.h>` at `opennurbs_system.h:389`
2. Had to copy all netpbm headers from `/opt/homebrew/include/netpbm` to `brlcad/include`
3. Had to specify `cmake -DCMAKE_CXX_FLAGS="-O2" ..`  because mismatching optimization for PCH of `CONFIG_CONTROL_DESIGN_stepg`
4. Had to add following to `gdal_priv.h:98` because somehow macro `STRCASECMP` was not recognized properly.
```
            #ifdef __APPLE__
            return strcasecmp(a, b) < 0;
            #else
```
5. Warning complains over SHA1 mismatch.
```
CMake Warning at misc/CMake/BRLCAD_Util.cmake:102 (_message):
  Local checkout SHA1 (e8cdc58fb0724bd74e42513b9d75bac7343a03f8) does not
  match latest bext 'main' branch SHA1
  (8075a8544d959c0769aa7d51d9041ca6cfb1a99c).  Mistmatch typically indicates
  either upstream repo updated and you need to update your local clone
  manually, or the brlcad_bext_init() argument in top level CMakeLists.txt
  needs 8075a8544d959c0769aa7d51d9041ca6cfb1a99c.
```

Elastino2, 2025-08-21:
Created [https://github.com/BRL-CAD/brlcad/pull/196](url) to remove `@par` issues.