BRL-CAD/brlcad #77: MGED on OSX fails - ogl_open: Can't get an appropriate visual.

fancsali, 2023-04-30:
A freshly built BRL-CAD 7.32.6 will fail with an error, if I try to start `mged`:

```
$ bin/mged 
ogl_open: Can't get an appropriate visual.

```

It does open two XQuartz windows, but they are both empty (and rather small) and the CLI just hangs there.

Only way to quit is Ctrl-C.

gwbrown, 2023-07-16:
I'm having the same problem.

gwbrown, 2023-07-18:
I actually can't find any evidence anyone's been able to successfully compile this for MacOS since 7.24.0. Did MacOS support get dropped from this project?

starseeker, 2023-09-09:
I am able to build BRL-CAD on OSX, at least with the latest main branch sources - you'll probably need -DENABLE_ALL=ON though, as the system Tcl/Tk won't work.