BRL-CAD/brlcad #83: itcl_methods.c:1159:48: error: invalid application of ‘sizeof’ to incomplete type ‘char[]’

iamaer4fa, 2023-07-17:
(Disclaimer: Not a C++ programmer)
My environment: Debian 12 (64-bit)
./configure is successful, but "make" has this error:

/brlcad-rel-7-34-2/src/other/ext/itcl3/generic/itcl_methods.c:1159:48: error: invalid application of ‘sizeof’ to incomplete type ‘char[]’
 1159 |         (unsigned)(sizeof(CompiledLocal)-sizeof(localPtr->name) + nameLen+1)

iamaer4fa, 2023-07-17:
Google's Bard suggested this code fix:

`localPtr = (CompiledLocal*)ckalloc(
        sizeof(CompiledLocal) + nameLen + 1
    );`
And it appears it fixes the issue.

starseeker, 2023-09-09:
I think a163442234b1092c858ffb3d6339972d0f8b1b73 handles this issue - let us know if there's still a problem.

closed 2023-09-09 by starseeker