BRL-CAD/brlcad #85: Build error [Linux] error: writing 1 byte into a region of size 0

kevinsmia1939, 2023-07-21:
Hi,
I am trying the build BRL-CAD from source.
I use openSUSE Tumbleweed.
I encounter the error below when running `make -j6`
Thank you for your help.

```
git clone https://github.com/BRL-CAD/brlcad
mkdir build
cd build
cmake .. -DBRLCAD_BUNDLED_LIBS=ON -DCMAKE_BUILD_TYPE=Release
make -j6
```

```
        Prefix: /usr/brlcad/rel-7.34.3
      Binaries: /usr/brlcad/rel-7.34.3/bin
     Libraries: /usr/brlcad/rel-7.34.3/lib
  Manual pages: /usr/brlcad/rel-7.34.3/share/man
Data resources: /usr/brlcad/rel-7.34.3/share
 
CC       = /usr/bin/cc
CXX      = /usr/bin/c++
CFLAGS   = -std=c11 -D_POSIX_C_SOURCE=200809L -D_XOPEN_SOURCE=700 -pipe
           -fvisibility=hidden -fno-strict-aliasing -fno-common -fexceptions
           -m64 -g -ggdb3 -O3 -fipa-pta -fstrength-reduce
           -fexpensive-optimizations -finline-functions -flto
           -fno-omit-frame-pointer -pedantic -pedantic-errors -Wall -Wextra
           -Wundef -Wfloat-equal -Wshadow -Wbad-function-cast -Wc++-compat
           -Winline -Wno-long-long -Wno-variadic-macros -Werror
CXXFLAGS = -std=c++14 -D_POSIX_C_SOURCE=200809L -D_XOPEN_SOURCE=700 -pipe
           -fvisibility=hidden -fno-strict-aliasing -fno-common -fexceptions
           -ftemplate-depth-128 -m64 -g -ggdb3 -O3 -fipa-pta -fstrength-reduce
           -fexpensive-optimizations -finline-functions -flto
           -fno-omit-frame-pointer -pedantic -Wall -Wextra -Wundef
           -Wfloat-equal -Wshadow -Wno-inline -Wno-long-long
           -Wno-variadic-macros -Werror
LDFLAGS  = -m64 -g -ggdb3
 
 
Compile Tcl ........................: ON
Compile Tk .........................: Disabled
Compile Itcl/Itk ...................: ON (Itcl only)
Compile Iwidgets ...................: OFF
Compile libpng .....................: ON
Compile libregex ...................: ON
Compile zlib .......................: ON
Compile STEPcode....................: ON
 
OpenGL support (optional) ..........: OFF
X11 support (optional) .............: OFF
Qt support (optional) ..............: OFF
Run-time debuggability (optional) ..: ON
 
Build 32/64-bit release ............: 64BIT (Auto)
Build with optimization ............: ON
Build static libraries .............: ON
Build dynamic libraries ............: ON
Install example geometry models ....: ON
Generate extra docs ................: ON (html/man)
```
```
cmake version 3.26.4
gcc (SUSE Linux) 13.1.1 20230710 [revision f9781fcf733ca3afe5b5811afd8ecbb18d9d3a11]
```

```
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/src/libbu/bitv.c: In function ‘bu_bitv_and’:
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/src/libbu/bitv.c:172:16: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
  172 |         *out++ &= *in++;
[... 88 lines cut ...]
  149 |         *out++ |= *in++;
      |                ^
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/include/bu/bitv.h:111:12: note: at offset [20, 2305843009213693948] into destination object ‘bits’ of size 2
  111 |     bitv_t bits[2];     /**< variable size array */
      |            ^
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/include/bu/bitv.h:111:12: note: at offset [4, 12] into destination object ‘bits’ of size 2
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/include/bu/bitv.h:111:12: note: at offset [20, 2305843009213693948] into destination object ‘bits’ of size 2
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/src/libbu/bitv.c:149:16: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
  149 |         *out++ |= *in++;
      |                ^
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/include/bu/bitv.h:111:12: note: at offset [21, 2305843009213693949] into destination object ‘bits’ of size 2
  111 |     bitv_t bits[2];     /**< variable size array */
      |            ^
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/include/bu/bitv.h:111:12: note: at offset [5, 13] into destination object ‘bits’ of size 2
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/include/bu/bitv.h:111:12: note: at offset [21, 2305843009213693949] into destination object ‘bits’ of size 2
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/src/libbu/bitv.c:149:16: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
  149 |         *out++ |= *in++;
      |                ^
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/include/bu/bitv.h:111:12: note: at offset [22, 2305843009213693950] into destination object ‘bits’ of size 2
  111 |     bitv_t bits[2];     /**< variable size array */
      |            ^
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/include/bu/bitv.h:111:12: note: at offset [6, 14] into destination object ‘bits’ of size 2
/home/kevin/Desktop/Flatpak/org.brlcad.BRL_CAD/brlcad/include/bu/bitv.h:111:12: note: at offset [22, 2305843009213693950] into destination object ‘bits’ of size 2

```

kevinsmia1939, 2023-07-21:
Adding -Wno-error=stringop-overflow as a temporary work around for now.

starseeker, 2023-09-09:
@brlcad I've seen this crop up too during some compiles - should the bits array be a variable sized array, i.e. bitv_t *bits?  The comment is pretty strange given the bits array isn't in fact variable sized...