BRL-CAD/brlcad #98: BRL-CAD_7.36.0_win64.exe Blocked by Antivirus

millercr4, 2023-08-30:
When attempting to download BRL-CAD_7.36.0_win64.exe, the download was blocked due to detected Trojan:Win32/Wacatac.H!ml. This did not occur when downloading the .msi. The BRL-CAD 7.36.0 release page shows the that .exe was updated Aug 14th, 2023 but the other release assets show Jul 26th and 27th.

starseeker, 2023-08-30:
We had gotten a previous report on the .exe installer, and I regenerated it with the latest NSIS from sourceforge.  Is there any way to confirm whether there could be a false positive?

bckelley, 2023-10-03:
I just had the same issue with the msi

timbolin, 2023-10-11:
i also just had this with the exe. id really appreciate if someone could confirm whether or not this was a false positive. vitustotal has a LOT of detections for it when analyzed so im a bit concerned.

starseeker, 2023-10-11:
I'm not sure what is triggering the reports... I've taken down the .exe for now, since the .msi seems to be causing fewer issues.  I'll try regenerating the .exe again later.

starseeker, 2023-10-11:
I thought it might be NSIS tripping up the scanners, but submitting a zip file of freshly compiled BRL-CAD build outputs to VirusTotal still results in some vendors flagging some of the executable files.  I'm thinking false positive based on it being a fresh compile without going through NSIS.  Six of them don't seem to like coil.exe, of all things...

starseeker, 2023-10-11:
OK, yeah...   Building with the Windows development VM from https://developer.microsoft.com/en-us/windows/downloads/virtual-machines and not using any installers (just making a zip file of the build) VirusTotal still returns a bunch of flags.