BRL-CAD/brlcad #100: unable to suppress bu_log from multiple threads simultaneously

kennethsdavis, 2023-10-25:
I am using the BRL-CAD API in a multi-threaded application and would like to suppress all log messages printed to the screen by bu_log. Calling bu_log_hook_delete_all and bu_log_add_hook with an empty function is a near solution. However, if one thread is inside the call to log_call_hooks when a second thread calls bu_log, then the second thread will have its message printed to the screen.

Providing a function that silences bu_log would solve the problem. Though it might be nice to make bu_hook_call thread-safe.

kennethsdavis, 2023-10-25:
To reproduce the error on windows build and run the following code.
*This is untested on Linux
```
#include <thread>
#include <vector>

#include <brlcad.h>

void thread_function()
{
   while(true)
   {
      bu_log("bu_log: thread log message\n");
      std::this_thread::sleep_for(std::chrono::milliseconds(100));
   }
};

int main(int argc, char** argv)
{
   bu_log("bu_log: test log message\n");

   // after this call to bu_log_add_hook no calls to bu_log should result in a
   // message being printed
   bu_log_add_hook([](void*,void*) { return 0; }, NULL);

   std::vector<std::thread> threads;

   // set thread_count to 1 to verify that bu_log doesn't print any messages in
   // a single-threaded context
   const unsigned int thread_count = std::thread::hardware_concurrency();
   std::generate_n(std::back_inserter(threads), thread_count, []() { return std::thread(thread_function); });

   // because the thread_function has an infinite loop the only way to quit is
   // to CTRL-C your way out of the application. all remaining code is
   // effectively useless.
   std::for_each(std::begin(threads), std::end(threads), [](auto& thread) { thread.join(); });

   return EXIT_SUCCESS;
}
```

AnkushSingh0710, 2023-12-04:
hello i'm a beginner, can you please assign me some beginner level issues or any bugs.

brlcad, 2023-12-04:
@AnkushSingh0710 Since you're totally commenting on this thread seemingly randomly -- you could try to address this issue.

@kennethsdavis Thank you for the report.  I'd be very interested to hear more about this multithreaded application of yours.  That said, thank you very much for the code snippet that demonstrates this issue and the detailed writeup.  I agree that pattern should work the way you're calling without erroneous logging and is quite unexpected.  I'm guessing we don't see it in our own applications because logging is not suppressed by default.  There certainly isn't a unit test for it (as yet).

kennethsdavis, 2023-12-05:
I’m working on an impact simulation. We’re using the brlcad API to fire thousands of rays through models to characterize the material overlap between them. All this can happen from multiple perspectives at once. It’s been an adventure getting this to work.

kennethsdavis, 2023-12-07:
@brlcad How do I get this issue assigned to me?

brlcad, 2025-08-04:
FYI, @kennethsdavis as you know your PR change was incorporated a long while ago, but there was an oversight giving you proper attribution in our release notes.  The mistake was noticed recently, so you will be recognized for the improvement in our upcoming 7.42 release.  Thank you again for the fix and careful test case!

If you have any public screenshots or publications of the impact simulation you mentioned, that would be fun to see.  If they are not public, feel free to reach out on other channels.

closed 2025-08-04 by brlcad

kennethsdavis, 2025-08-06:
@brlcad, the only thing I have that is publicly releasable is more program-related than software-related. If you're interested I can send it. What's the best way to get it to you?