BRL-CAD/brlcad #111: Missing download release installers

luzpaz, 2024-01-06:
https://github.com/BRL-CAD/brlcad/releases/tag/rel-7-38-0 is missing assets

luzpaz, 2024-01-20:
Can anyone assist? 
@starseeker ? (sorry to ping)

starseeker, 2024-01-22:
Are you looking for  a Windows installer?

starseeker, 2024-01-22:
I've got a .exe installer up.

luzpaz, 2024-01-22:
Is there a reason why you're not providing linux binaries (as done in 7.36.0) ?

[image]

starseeker, 2024-01-23:
The building of binaries is currently a manual process, and it's not always clear how much demand there is for the Linux binaries vs. people compiling specifically for their own systems.  Historically it was more straightforward to compile on Linux than Windows (although that's less true these days) and we've not put a lot of focus into figuring out all the subtleties of making a Linux binary that works well across lots of systems (Xorg and glibc compatibility concerns, for example.)  Usually I build those on a CentOS or Fedora VM, but those aren't my daily use systems so it's usually not a priority.

If there is documented need/interest for the Linux binaries from users then that ups the incentive to make more of an effort to produce them regularly.

Dragon1573, 2024-01-23:
Is it able to configure CI/CD workflows (such as GitHub Workflows, Travis CI, Circle CI) for automatically compiling and building releases assets when pushing a release tag? I think it may be convenient for you official members, as you don't have to manually compile, upload and release them.

Kristof64, 2024-01-23:
> If there is documented need/interest for the Linux binaries from users then that ups the incentive to make more of an effort to produce them regularly.

I for one would appreciate binaries for Linux 😊

starseeker, 2024-01-23:
> Is it able to configure CI/CD workflows (such as GitHub Workflows, Travis CI, Circle CI) for automatically compiling and building releases assets when pushing a release tag?

Probably that is possible, but I don't have enough expertise with the Actions system to set that up myself at the moment.  If someone wants to contribute to the project figuring out how to set that up would be an excellent opportunity to do so.

luzpaz, 2024-11-15:
Anyone want to take point on this ?