BRL-CAD/brlcad #122: issue compiling "rle.h"

bralani, 2024-04-02:
There is an issue related to this commit  https://github.com/BRL-CAD/brlcad/commit/3d3fc7ccbe06e99b4d0a659fe8381195c1c37f2b

 When I compile the project, the build fails because:

Error C1083 Cannot open include file: 'rle.h': No such file or directory D:\brlcad\build\Debug\brlcad D:\brlcad\src\libicv\rle.c on line 28

It seems there is no "rle.h" file in the same folder of "rle.c", I don't know if I am not doing properly the build...

@starseeker

starseeker, 2024-04-03:
I think b412dcb may have fixed this issue - is it still broken for you with that change?

bralani, 2024-04-03:
> I think [b412dcb](https://github.com/BRL-CAD/brlcad/commit/b412dcbd866828d1d89e5e6dee9d05afae946768) may have fixed this issue - is it still broken for you with that change?

Ah ok now it works. It seems I had cloned the repo before that fix. Thanks.

closed 2024-04-03 by bralani