BRL-CAD/brlcad #125: translate and rotate from text file.

bentleyeglin, 2024-04-17:
I have a .g that was built from STL.  It has one region and the path is all/r.stl/s.stl
s.stl is not a solid, but rather a BOT Bag of Triangles.
I need to be able to translate and rotate this region from a text file: mged model.g < input.txt
sed does not work.
I have tried:
sed s.stl
keypoint 0 0 0
orot 0 90 0
tra 100 0 0
accept
This and other variations of this does not work.  I think it may have to do with it being a BOT and not a solid.  Can anyone help me out?

brlcad, 2024-04-17:
I was going to say, you have to draw/e the object before you can sed it.  After doing that, either rot or orot should work.

brlcad, 2024-04-17:
Also just a note -- it's only useful to set a keypoint for rotations.  It tells it which point to rotate about.  For translations it doesn't hurt, it just doesn't matter though either.

closed 2024-04-17 by brlcad