BRL-CAD/brlcad #130: Faster way to build large amount of spheres.

bentleyeglin, 2024-05-22:
I'm using BRL-CAD to plot spheres.  I have lots and lots of spheres.  I know there are tools made for plotting, but I have a case where I am placing spheres on an existing model.  I don't want to expand on why, but let's say it makes for great pictures when you raytrace.
I am currently using a text file to input each sphere and it takes a long time.  I'm doing: mged model.g < text.txt

in sph01.s sph 0 0 0 1
r sph01.r u sph01.s
g spheres sph01.r
comb_color sph01.r 254 0 0

This can get really slow if I have tens of thousands of spheres.

brlcad, 2024-05-22:
No need to explain, I've struggled recently with this very same issue!
[image]
[image]
[image]

For what it's worth, the main bottleneck is mged's display management.  It gets linearly slower as you keep drawing more and more entities.  **One trick you can try if you don't have it yet is to insert a "Z" command after every set.**

Calling the "in" command causes mged to draw the primitive (a 'sph' in your case, but same issue for other entity types too), and as you call it over and over, the list of things needing to get redrawn grows.  Calling the Z command erases everything, and thus avoids the non-obvious growing redraw costs and speeding things up immensely.

That should get you drawing tens of thousands pretty efficiently.  If you need to go even bigger, say millions, you'll then also want to adjust your grouping strategy.  Instead of grouping everything into spheres, group them in sets (e.g., g spheres_a sph_a_01.r) and then group your sets (e.g., g spheres spheres_a spheres_b spheres_c ...). That way, you're not having to grow and reallocate a large group as you get into hundreds of thousands or millions.

Would love to see any pictures or details you are able to share, if not here through other channels.

last comment -- our upcoming 7.38.4 release has a change that improves the speed of mged scripting, but that's probably not going to be terribly beneficial to you until the other bottlenecks are addressed.

brlcad, 2024-05-24:
Please do let me know if that helped your performance issue... or if that didn't help.

bentleyeglin, 2024-05-24:
It absolutely helped.  Thank you very much.

David Bentley
EPASS Support Contractor
***@***.***
www.axientcorp.com<http://www.axientcorp.com/>

Digital Acquisition & Sustainment Office| Modeling, Simulation and Analysis
AFLCMC | EBRD
Eglin Air Force Base, Florida
DESK: 850-883-4513
CELL: 850-499-6609, call or text
***@***.******@***.***>
***@***.******@***.***>