BRL-CAD/brlcad #148: Help with nirt command.

bentleyeglin, 2024-07-16:
I have read the nirt documentation and it makes no sense to me.  Maybe someone with more knowledge can help.
I would like to feed nirt an xyz origination point and an ae (azimuth and elevation)  and have nirt return if it hit the geometry and what the xyz is where it intersected.  I'm going to script this to feed it a file containing  x y z a e.

Is this possible with nirt or is there a better way?

bentleyeglin, 2024-07-24:
This one took me a while to get, but I finally got it.  The manual was just not clicking with me.  Anyway, to fire a nirt at a target using x,y,z and Az,El I found the following command to work:
nirt -b -s -e "xyz $x $y $z; ae $az $el; s; q" model.g all

ahanachauhann, 2025-01-24:
Yes, it's possible to use nirt to fire a ray from a given origin (x, y, z) with azimuth (a) and elevation (e) angles and check for intersections with geometry.Convert Azimuth and Elevation to Direction Vector,then Run nirt with these values : "nirt -b -s -e "xyz $x $y $z; ae $az $el; s; q" model.g all"  If you're feeding a file with multiple coordinates and azimuth values, you can loop through the file and run the above command for each entry then you can check for the output.