BRL-CAD/brlcad #152: Bot Raytracing Anomaly

cmpitts99, 2024-07-31:
I have found a possible anomaly with bot raytracing. I have replicated this in both Windows and Linux, using versions 7.26.4 and 7.38.2.
 
I created a simplified .g file with two plate mode bots and two rpps.

bot1.s has 3 points and 1 triangle  (0 0 5) (0 5 -5) (0 -5 -5) thickness = 20, centered about hit point
bot2.s has 3 points and 1 triangle  (20 0 5) (20 5 -5) (20 -5 -5) thickness = 20, centered about hit point
rpp1.s is from -20 to 20, 5.1 to 25.1, -20 to 20
rpp2.s is from -20 to 20, -25.1 to -5.1, -20 to 20

group1 contains bot1.r
group2 contains bot1.r and bot2.r
group3 contains bot1.r rpp1.r rpp2.r
group4 contains bot1.r bot2.r rpp1.r rpp2.r

When shooting a ray down the negative x axis (-1,0,0), group1 will register a hit down to position -1,0,0
When shooting a ray down the negative x axis (-1,0,0), group2 will register a hit down to position -1,0,0
When shooting a ray down the negative x axis (-1,0,0), group3 will register a hit down to position -2,0,0
When shooting a ray down the negative x axis (-1,0,0), group4 will register a hit down to position -9.99,0,0

I have tested several thicknesses for bot1.s and group4 seem to always register a hit just short of the full bot thickness on the negative side. 
[image]

brlcad, 2024-08-01:
@cmpitts99 Thanks for the report.  Can you attach your .g file?

cmpitts99, 2024-08-01:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

cmpitts99, 2024-08-01:
[TestBot.g.gz](https://github.com/user-attachments/files/16456023/TestBot.g.gz)

starseeker, 2024-09-26:
I did try to reproduce this but wasn't able to (at least on Linux) - what version of Linux did you see this on?

cmpitts99, 2024-09-26:
Rocky 8

cmpitts99, 2024-09-26:
I can recreate in mged:
e group1
ae 0 0
nirt  -1 0 0  (Hit)
nirt  -2 0 0  (Miss)

e group2
ae 0 0
nirt  -1 0 0  (Hit)
nirt  -2 0 0  (Miss)

e group3
ae 0 0
nirt  -1 0 0  (Hit)
nirt  -2 0 0  (Hit)
nirt  -3 0 0  (Miss)

e group4
ae 0 0
nirt  -9 0 0        (Hit)
nirt  -9.99 0 0   (Hit)
nirt  -10 0 0     (Miss)