BRL-CAD/brlcad #154: Compile Error:  lto1: fatal error

Crtomir4, 2024-08-13:
I get the following error when trying to compile BRL-CAD on Ubuntu 22.04 LTS with gcc version 11.4.0 using the "make" command.  It seems time64.c.o was generated with a different version of LTO than the one that is being used for compilation. Any ideas how to fix this?

'
[ 17%] Built target libfft
[ 17%] Built target qged_plugins
[ 18%] Built target libfft-static
[ 26%] Built target docbook-docbook-system-mann
[ 26%] Built target fftest
lto1: fatal error: bytecode stream in file ‘CMakeFiles/libbu-obj.dir/y2038/time64.c.o’ generated with LTO version 11.3 instead of the expected 12.0
compilation terminated.
lto-wrapper: fatal error: /usr/bin/c++ returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[2]: *** [src/libbu/CMakeFiles/libbu.dir/build.make:273: lib/libbu.so.20.0.1] Error 1
make[1]: *** [CMakeFiles/Makefile2:13990: src/libbu/CMakeFiles/libbu.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
'

Ravish990, 2024-12-14:
can you please assign the issue ? I would love to fix it and contribute to the organisation.

Lakshitalearning, 2025-03-13:
The object file time64.c.o was built with LTO version 11.3, but in your case your current compiler expects LTO version 12.0 (from gcc 11.4.0).
You can try possible fixes if still the issue persist: 

**1. If LTO isn't necessary , you can try disabling it temporarily using below code in bash:**
```
cmake -DCMAKE_C_FLAGS="-fno-lto" -DCMAKE_CXX_FLAGS="-fno-lto" ..
make 
```

**2. Sometimes .o (object files) which are leftover and exist from other compiler can bring up this issue. You can try removing these .o files and make a clean build:**

`make clean
rm -rf build
mkdir build && cd build
cmake ..
make -j$(nproc) `

**3. Confirm that the same GCC version is used across all steps. Set the CC and CXX variables explicitly to enforce the correct compiler.**

If the issue persists, it may be helpful to check for any cached CMake settings or investigate deeper LTO-related flags.