BRL-CAD/brlcad #157: CMAKE_MAKE_PROGRAM should be forwarded to ExternalProject_Add()

bylee20, 2024-08-29:
In CMakeLists.txt scripts from bext, ExternalProject_Add() missing CMAKE_MAKE_PROGRAM option.
I think it should be forwarded likewise CMAKE_CXX_COMPILER and CMAKE_C_COMPILER.
For instance, I am using ninja to build on Windows and configure cmake with CMAKE_MAKE_PROGRAM=FILEPATH:path_to_ninja.
There's no way to forward this option to ExternalProject_Add(), all subsequence build process will fail for unknown make program.

starseeker, 2024-09-01:
Hmm.  That should be doable, although we'd need to make sure doing that works if it is not explicitly set.  I don't think any of the current devs have ever used that variable to specify a make program, so it wasn't on our radar.

If you've already made the changes locally, could you submit a pull request to the bext repository?  If not I can look into it, but it may be a little bit due to other priorities.

bylee20, 2024-09-03:
I have made changes but it's only for minimal build.
I can write a pull request to add CMAKE_MAKE_PROGRAM to all ExternalProject_Add() which looks it is required, but I cannot test it because it is hard to try building the entire dependencies.
If you don't mind the absence of test, I will make a PR.

starseeker, 2024-09-05:
That would be helpful  - thanks!

bylee20, 2024-09-19:
I've noticed that it is way easier to use default generator than to patch all `CMakeLists.txt` for ninja and I don't need this for now.
Thank you for your consideration.

closed 2024-09-19 by bylee20