BRL-CAD/brlcad #159: /utf-8 flag should be set for MSVC build

bylee20, 2024-08-29:
some files, for instance, fort.c require UTF-8 encoded source code support which is disabled on MSVC by default.
You should append /utf-8 flag to enable UTF-8 source code support on MSVC.

starseeker, 2024-09-01:
Interesting.  What version of Visual Studio are you using to build?  I don't think I've seen a UTF-8 build failure myself with MSVC...

bylee20, 2024-09-03:
> Interesting. What version of Visual Studio are you using to build? I don't think I've seen a UTF-8 build failure myself with MSVC...

It is VS2022 but I don't think this is version specific problem.\
/utf-8 flag exists since VS2017 at least and it might be there in VS2015 too.

Maybe your codepage is compatible with UTF-8?
My codepage is CP949 which is not compatible with UTF-8 and MSVC uses the OS' codepage when dealing with encoding.

starseeker, 2024-09-09:
Can you let us know if https://github.com/BRL-CAD/brlcad/commit/8012562f1d2445bbe4a00f608ff9467e7958e1f6 fixes your issue?

bylee20, 2024-09-19:
Yes, thank you!

closed 2024-09-19 by bylee20