BRL-CAD/brlcad #161: rt_db_put_internal() does not work with inmem database

bylee20, 2024-09-19:
I have created empty database with db_create_inmem() and It seems that `rt_db_put_internal()` does not work with inmem database.
Here's my test code:

```
    const auto db = db_create_inmem();

    rt_ell_internal *sph = nullptr;
    BU_ALLOC(sph, rt_ell_internal);
    *sph = {
        .magic = RT_ELL_INTERNAL_MAGIC,
        .v = {0, 0, 0},
        .a = {1, 0, 0},
        .b = {0, 1, 0},
        .c = {0, 0, 1},
    };

    rt_db_internal in = {
        .idb_magic = RT_DB_INTERNAL_MAGIC,
        .idb_major_type = DB5_MAJORTYPE_BRLCAD,
        .idb_minor_type = ID_SPH,
        .idb_meth = &OBJ[ID_SPH],
        .idb_ptr = sph,
        .idb_avs = BU_AVS_INIT_ZERO,
    };

    const auto dir = db_diradd(db, "sph.o", RT_DIR_PHONY_ADDR, 0, RT_DIR_SOLID, &in.idb_minor_type);
    rt_db_put_internal(dir, db, &in, &rt_uniresource);
```

If you execute this code, you will see error message:

> rt_db_put_internal5(sph.o) db_realloc5() failed

I did short grepping and found that this error comes from:
https://github.com/BRL-CAD/brlcad/blob/02ba341f2ea0c58c896a9224ec8c6ed7b434f6db/src/librt/db5_io.c#L767-L773

which is result of failure of `db5_realloc()` call and again, this failure is originated from:
https://github.com/BRL-CAD/brlcad/blob/02ba341f2ea0c58c896a9224ec8c6ed7b434f6db/src/librt/db5_alloc.c#L135-L141

Here, `db_dirbuild()` call always fails for inmem database because of this check:
https://github.com/BRL-CAD/brlcad/blob/02ba341f2ea0c58c896a9224ec8c6ed7b434f6db/src/librt/db5_scan.c#L387-L389

Since dbip is for inmem database, the file pointer will be always null and db_dirbuild() never work.

I also found that wdb_put_internal() works fine with inmem database.

Is this by design or a bug? Or, did I do something wrong?

starseeker, 2024-09-26:
@brlcad do you have any insights about the design intent here?  Is rt_db_put_internal expected to work in this case?

bylee20, 2024-09-30:
> @brlcad do you have any insights about the design intent here? Is rt_db_put_internal expected to work in this case?

Thank you for your attention.

I think most people would expect it will work unless it is explicitly documented, since... I mean, literally, why not?

As a matter of fact, I found this while using libged with inmem db.
It seems that libged can work with inmem db by giving `"inmem"` and the address of `db_i` pointer as arguments.
However, as soon as I attempted to make a geometry with `"make"` command, this failed because of this.
As I mentioned in the issue, I can use `wdb_put_internal()` but this is not possible when interop with libged because the `rt_db_put_internal()` call is done in libged side.

brlcad, 2024-09-30:
@bylee20 it's basically a bug.  As you note, it could/can work, so why not.  It should work.  The practical issue is just that all the API code paths (and there's dozens of paths for this particular one) for reading and writing in-mem databases were narrowly developed for specific features, so more work is needed in other areas that are not yet exercised by an end-user feature.

closed 2026-09-16 by starseeker