BRL-CAD/brlcad #162: Primitive Selector Issue with Debian 12 and Primitive Selector    BRL-CAD-7-40-2

jm684, 2024-10-16:
[image]

Hi all,

I am learning BRL CAD on Debian 12, after downloading from Github.

As you can see, in Screen shot, i drew Rectangle1.s, Cylinder1.s and Sphere1.s.

After drawing, Bar Menu: Edit, Primitive Selection and no result.

Would you kind enough to tell me, if if forgot some step or misunderstanding ? , in order to use  this  soft.

This basic exercise is only to check Primitive Selection, after same result with Goblet.g exercise.

Thanks a lot, and nice day.

jm

brlcad, 2024-10-16:
Hi jm.  You're not doing anything wrong.  We discovered that bug had made its way into the 7.40 release just recently.  It's been fixed in our latest git sources, but we've not yet posted a source release with the fix.  I believe the 7.38 release doesn't have the issue, which affects both the Primitive Edit and Matrix Edit GUI panels in MGED.  

As a workaround, you can run the "sed" command on the command line instead of using the Primitive Edit GUI panel.  Just draw your object (e.g., "draw Sphere1.s") and then run sed aka solid edit on it (e.g., "sed Sphere1.s").  From there, you can accept/reject edits as usual via the Edit menu.

jm684, 2024-10-16:
Hi Christopher,I use now BRL-CAD 7.36, works fine.Thanks for your fast reply.That's why i only use in my own pc, opensource softs.😃Congratulations for your great team work.Learning curve is steep,  but worth to do.Nice Day.Jean-michel

jm684, 2024-10-16:
Christopher, Hi again,7.38 has the same issue, that's why i use 7.36....Thanks from France.