BRL-CAD/brlcad #165: Anyone get BRL-CAD approved on NIPRNET?

bentleyeglin, 2024-10-22:
I'm trying to get BRL-CAD installed on my NIPR machine and my IT is wary of the "open source" aspect of the tool.  I'm going to begin the process of requesting it be installed, but if anyone has been there and done that, I would appreciate any insight.

-David

brlcad, 2024-10-22:
@bentleyeglin If you reach out via DSIAC https://dsiac.org/models/ajem/ and/or via gov't e-mail to someone at DEVCOM, BRL-CAD can be requested and/or sent over DoD SAFE.  That typically appeases IT as it is like any other GOTS application.  Installer will functionally be 100% identical to the GitHub download, but built and deployed such as to ensure compatibility and facilitate compliance.  BRL-CAD is on NIPR and SIPR across various organizations without issue.

On a related technical note, some find it convenient to discover BRL-CAD's installer does not require administrative privilege.  Installer just needs a place where files can be written.  That is intentional for users and system administrators alike.

bentleyeglin, 2024-10-22:
Thank you Sean.  I am coordinating with my IT and my DSIAC account holder.

closed 2024-10-22 by starseeker