BRL-CAD/brlcad #168: Linux compilation error bext/geogram

envolution, 2024-11-09:
I have also tried with release tarball 7.40.2 that had the same error

Arch Linux
gcc/g++ (GCC) 14.2.1 20240910
[cmake.log](https://github.com/user-attachments/files/17689268/cmake.log)

cmake version 3.31.0

CFLAGS="${CFLAGS} -lm" cmake \
    -S . \
    -G Ninja \
    -B "${srcdir}/build" \
    "-DCMAKE_INSTALL_PREFIX=/opt/brlcad" \
    "-DCMAKE_BUILD_TYPE=Release" \
    -DBRLCAD_BUNDLED_LIBS=ON \
    -DBRLCAD_ENABLE_STRICT=NO \
    -DBRLCAD_ENABLE_QT=ON 2>&1 | \
    tee cmake.log

```
stderr output is:
...skipping to end...
 "_FORTIFY_SOURCE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "_FORTIFY_SOURCE" redefined
...
<command-line>: warning: "_FORTIFY_SOURCE" redefined
<command-line>: note: this is the location of the previous definition
/tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/geogram/third_party/stb_image/stb_image.h: In function 'stbi__parse_png_file.constprop':
/tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/geogram/third_party/stb_image/stb_image.h:5166:56: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
 5166 |                   for (k = 0; k < s->img_n; ++k) tc[k] = (stbi_uc)(stbi__get16be(s) & 255) * stbi__depth_scale_table[z->depth]; // non 8-bit images will be larger
      |                                                        ^
/tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/geogram/third_party/stb_image/stb_image.h:5082:25: note: at offset 3 into destination object 'tc' of size 3
 5082 |    stbi_uc has_trans=0, tc[3]={0};
      |                         ^
make[3]: *** [Makefile:156: all] Error 2

CMake Error at /tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD-stamp/GEOGRAM_BLD-build-Release.cmake:47 (message):
  Stopping after outputting logs.

make[2]: *** [geogram/CMakeFiles/GEOGRAM_BLD-install.dir/build.make:83: geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD-stamp/GEOGRAM_BLD-build] Error 1
make[1]: *** [CMakeFiles/Makefile2:3397: geogram/CMakeFiles/GEOGRAM_BLD-install.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

starseeker, 2024-11-10:
Could you try building geogram itself as a stand-alone build?  If the error persists there (it should) then a bug report should be submitted to that project.  It looks like the stb_image header is where this is coming from, so the fix will probably be to either update geogram's stb_image or fix stb_image itself.

envolution, 2024-11-10:
BrunoLevy/geogram compiles ok on it's own.  I did do a diff compare against stb_image/stb_image.h for brlcad and geogram and they matched.  I should let you know I did create a patch for slacon.c(bext/geogram/geogram/src/lib/third_party/numerics/SUPERLU/slacon2.c) [slacon_idamax_isamax.patch.txt](https://github.com/user-attachments/files/17693045/slacon_idamax_isamax.patch.txt) before capturing the log.  This is the compilation log without the patch:

```
-- stderr output is:
/tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/third_party/numerics/SUPERLU/slacon2.c: In function 'slacon2_':
/tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/third_party/numerics/SUPERLU/slacon2.c:163:27: error: passing argument 2 of 'idamax_' from incompatible pointer type [-Wincompatible-pointer-types]
  163 |     isave[1] = idamax_(n, &x[0], &c__1);  /* j */
      |                           ^~~~~
      |                           |
      |                           float *
/tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/third_party/numerics/SUPERLU/slacon2.c:25:27: note: expected 'double *' but argument is of type 'float *'
   25 | extern int idamax_(int *, double *, int *);
      |                           ^~~~~~~~
make[5]: *** [src/lib/third_party/numerics/CMakeFiles/geogram_num_3rdparty.dir/build.make:7933: src/lib/third_party/numerics/CMakeFiles/geogram_num_3rdparty.dir/SUPERLU/slacon2.c.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:290: src/lib/third_party/numerics/CMakeFiles/geogram_num_3rdparty.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
In file included from /tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/geogram/image/image_serializer_stb.cpp:85:
/tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/geogram/third_party/stb_image/stb_image.h: In function 'int stbi__parse_png_file(stbi__png*, int, int)':
/tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/geogram/third_party/stb_image/stb_image.h:5166:56: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
 5166 |                   for (k = 0; k < s->img_n; ++k) tc[k] = (stbi_uc)(stbi__get16be(s) & 255) * stbi__depth_scale_table[z->depth]; // non 8-bit images will be larger
      |                                                  ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/brl/src/build/bext_build/geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD/src/lib/geogram/third_party/stb_image/stb_image.h:5082:25: note: at offset 3 into destination object 'tc' of size 3
 5082 |    stbi_uc has_trans=0, tc[3]={0};
      |                         ^~
make[3]: *** [Makefile:156: all] Error 2

make[2]: *** [geogram/CMakeFiles/GEOGRAM_BLD-install.dir/build.make:83: geogram/GEOGRAM_BLD-prefix/src/GEOGRAM_BLD-stamp/GEOGRAM_BLD-build] Error 1
make[1]: *** [CMakeFiles/Makefile2:3132: geogram/CMakeFiles/GEOGRAM_BLD-install.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

starseeker, 2024-11-12:
Hmm.  I'm surprised it compiles OK alone.  Can you try https://github.com/BRL-CAD/geogram/tree/888172e37ea4729469fa2c5dbbc15ba68b8c234e as a stand-alone build?  If that fails, I guess the next step will be figuring out what changed compared to BrunoLevy/geogram (or whether one of our build options in https://github.com/BRL-CAD/bext/blob/main/geogram/CMakeLists.txt is a problem.

envolution, 2024-11-13:
It compiled but with some caveats:
It was missing some third-party directories:
triangle, tetgen, HLBFGS - I took them from the 'main' branch
```
/tmp/geogram/src/lib/third_party/numerics/SUPERLU/slacon2.c: In function ‘slacon2_’:
/tmp/geogram/src/lib/third_party/numerics/SUPERLU/slacon2.c:163:27: error: passing argument 2 of ‘idamax_’ from incompatible pointer type [-Wincompatible-pointer-types]
  163 |     isave[1] = idamax_(n, &x[0], &c__1);  /* j */
      |                           ^~~~~
      |                           |
      |                           float *
/tmp/geogram/src/lib/third_party/numerics/SUPERLU/slacon2.c:25:27: note: expected ‘double *’ but argument is of type ‘float *’
   25 | extern int idamax_(int *, double *, int *);
      |                           ^~~~~~~~
make[2]: *** [src/lib/third_party/numerics/CMakeFiles/geogram_num_3rdparty.dir/build.make:7933: src/lib/third_party/numerics/CMakeFiles/geogram_num_3rdparty.dir/SUPERLU/slacon2.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1301: src/lib/third_party/numerics/CMakeFiles/geogram_num_3rdparty.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

after applying the slacon2 patch I referred to in my initial comment, the geogram branch you linked compiled fully.  I do have a github actions script that compiles brlcad for arch that exhibits the same problems if you see any value in that I could trim it down to reproduce the error for you.  I'm also happy to perform any tests you'd like locally.

starseeker, 2024-11-15:
Hmm.  Those third party directories are deliberately removed, since there are licensing incompatibilities, so that's probably just a case of needing to turn on the options to disable the features.  Maybe the failure is specific to the config we have turning off those features?  https://github.com/BRL-CAD/bext/blob/main/geogram/CMakeLists.txt#L29

envolution, 2024-11-15:
I followed the Quick Start section of the README which specified:
cmake ../bext -DENABLE_ALL=ON

which is probably why those were needed. 

I've spent some time to patch the errors I've seen in the bext submodules, and am busy working on compiling the main project compilation now.  The submodules that needed changes were:

bext/geogram/geogram/src/lib/third_party/numerics/slacon2.c
bext/tkhtml/tkhtml/src/htmlprop.c
bext/manifold/manifold/src/third_party/clipper2/include/clipper2/clipper.core.h
bext/tkhtml/tkhtml/src/htmltree.c
bext/itk/itk/generic/itk_archetype.c

[bext_patches.zip](https://github.com/user-attachments/files/17781375/bext_patches.zip)

I'd try pull request but I'm not super familiar with the submodule/cmake structure of the project, and I've not yet actually tested these patches anyways.

I'll follow up and close this with more information later today I hope, for now the initial error is not present and I did not need to patch it specifically (not clear to me why this is, but possibly related to one of the included patches)

envolution, 2024-11-18:
I've gotten stuck with a segfaulting btclsh which seems to be caused by some issue with opennurbs and handling empty lists in opennurbs_array.cpp.  I'm going to put a hold on this for a while as I'm a little out of my depths trying to make sense of the various libraries needing changes to compile on Arch.

closed 2024-11-18 by envolution