BRL-CAD/brlcad #169: bext/jpeg: windows unresolved external symbols

kennethsdavis, 2024-12-18:
BRL-CAD main @ 2275f6046a
CMake 3.31.2, generator=Visual Studio 16 2019

There are a lot of unresolved symbols that should be in the CRT.

Build log attached.
[JPEG_BLD-build-out.log](https://github.com/user-attachments/files/18184201/JPEG_BLD-build-out.log)

starseeker, 2024-12-19:
Hmm.  The references I'm seeing to similar issues online seem to suggest a possibility of toolchain configuration issues... If you try to build the jpeg library in isolation, not as part of bext, does it succeed?

starseeker, 2024-12-20:
Is there any behavioral difference between a Debug build and Release build?  The next deeper step beyond that would be to inspect the build flags and toolchains being used in the isolated bext vs the configure-driven bext and try to see where (beyond paths, obviously) the differences lie.