BRL-CAD/brlcad #173: Make failed (debian)

SBDushenkov, 2025-02-02:
[build.log](https://github.com/user-attachments/files/18633087/build.log)

starseeker, 2025-02-04:
Can you tell us which version this is?  Latest main?

The failure is in OpenNURBS, which is one of the dependencies from https://github.com/BRL-CAD/bext - could you try building bext separately and seeing what the OpenNURBS build logs tell you?  Also, do you have libuuid-dev on your system?

SBDushenkov, 2025-02-19:
Steps to reproduce (for initial issue)

docker run  --rm --net=host  -it --entrypoint bash debian:latest
apt update
apt upgrade
apt install build-essential make cmake
apt install libc6-dev libfreetype-dev libfontconfig-dev
apt install xserver-xorg-dev libx11-dev libxi-dev libglu1-mesa-dev
apt install git
git clone https://github.com/BRL-CAD/brlcad.git
cd brlcad/
mkdir build
export BRLCAD_ENABLE_STRICT=OFF
export BRLCAD_ENABLE_COMPILER_WARNINGS=OFF
export BRLCAD_BUNDLED_LIBS=ON
cd build/
cmake .. -DBRLCAD_ENABLE_STRICT=NO -DBRLCAD_BUNDLED_LIBS=ON -DCMAKE_BUILD_TYPE=Release

And yes. Issue is related to BRL-CAD/bext
Make bext also failed

same preparation +
sudo apt-get install tk
git clone https://github.com/BRL-CAD/bext
mkdir bext_build && cd bext_build
cmake ../bext -DENABLE_ALL=ON -DCMAKE_BUILD_TYPE=Release
cmake --build . --config Release --parallel 8

starseeker, 2025-02-26:
In the bext build directory there should be a folder:

opennurbs/OPENNURBS_BLD-prefix/src/OPENNURBS_BLD-stamp/

with a bunch of log files for the various steps. Does any of them have a more informative error message?