BRL-CAD/brlcad #174: How to determine ARB6 index mapping?

bylee20, 2025-02-04:
It seems there are two schema to map index of vertices between ARB8 and ARB6:

* 0 1 2 3 4 4 5 5 
* 0 1 2 3 4 5 5 4

How can I figure out which one should I use from vertices?

bylee20, 2025-02-04:
It seems there exists another case aside 0/1/2/3/4/4/5/5 and 0/1/2/3/4/5/5/4.
I found that `s.bld10.o1` in `havoc.g` model has weird point set:

```
[0] 10358.09375 -80 3070.118896484
[1] 10360.885709762573 -345 3150.0701599117297
[2] 10360.885709762573 -345 3150.0701599117297
[3] 10360.885709762573 -80 3150.0701599117297
[4] 3081.83203125 -80 3304.1994323727
[5] 3084.62109375 -345 3384.1506347652
[6] 3084.62109375 -345 3384.1506347652
[7] 3084.62109375 -80 3384.1506347652
```

As you can see, duplicated points exist [1]/[2] and [5]/[6]. 
It seems that this scheme is not supposed to exist because, for instance, `l s.bld10.o1` shows only first 6 points,

```
Archer> l s.bld10.o1
s.bld10.o1:  ARB6
	1 (1035.81, -8, 307.012)
	2 (1036.09, -34.5, 315.007)
	3 (1036.09, -34.5, 315.007)
	4 (1036.09, -8, 315.007)
	5 (308.183, -8, 330.42)
	6 (308.462, -34.5, 338.415)
```

So, there's no way to edit the last point at [7].
The screen also shows non-annotated point:

[image]

Anyway, is there systematic way to extract unique points and determine index map?
Even though archer does not properly handles points, the drawn geometry seems correct and no issue with rt, so I think it is correctly modeled.

Btw,  is it possible to debug this kind of geometry in archer? I extracted the full point list using C-API but it would be nice to have such a command to check it in archer.

brlcad, 2025-02-04:
Excellent and timely question because I’m in the middle of working on the arb8 code, to extend it with support for concave definitions and noted some of the inconsistency you noticed.  Not all of them though!  I’ve not noticed the issue on havoc.  I’ll have to dig a bit before answering to make sure you get the right info..  

That said, are you working on code or in userland?

bylee20, 2025-02-07:
When you saying working on code, if you asking if I am using C-API of BRL-CAD's library, then yes I am working on code.