BRL-CAD/brlcad #177: Error during make

TheSallust, 2025-02-25:
[ 51%] Built target OPENNURBS_BLD-install
gmake: *** [Makefile:91: all] Error 2
CMake Error at misc/CMake/BRLCAD_Util.cmake:102 (_message):
  Unable to successfully build bext dependency repository
Call Stack (most recent call first):
  misc/CMake/BRLCAD_EXT_Setup.cmake:446 (message)
  misc/CMake/BRLCAD_ExternalDeps.cmake:529 (brlcad_ext_setup)
  CMakeLists.txt:2096 (brlcad_bext_process)

starseeker, 2025-02-26:
In the bext build directory there should be a folder:

opennurbs/OPENNURBS_BLD-prefix/src/OPENNURBS_BLD-stamp/

with a bunch of log files for the various steps.  Does any of them have a more informative error message?

TheSallust, 2025-02-27:
I do not have the folder OPENNURBS_BLD-prefix or OPENNURBS_BLD-stamp

starseeker, 2025-02-28:
Hmm.  Looking more closely, I see it thinks it built the install target for OpenNURBS.  Does the log report which component specifically failed?  (Guessing TCL, but might be something else...)

brlcad, 2025-02-28:
@TheSallust can you provide the entire cmake/build log?  that will help sort out what failed exactly.

agaztec, 2025-07-17:
Hi, I'm getting this exact same error. OpenNURBS(?) failing, building on M3 Mac on Sequoia 15.5. Funnily enough, I successfully performed the CMake step on OS 14, but then I was failing the GNU make step due to clang somehow not recognizing a datatype in its own malloc.h file, so I did a system update and started from the beginning (deleted the build folder) to make sure everything would still work on my system. Now, though, CMake is failing instead. 

Where exactly is the build log you're looking for? I have a CMakeOutput.log file in brlcad/build/CMakeFiles, but it makes no mention of OpenNURBS. The final block in the output is as follows:
```
***********************************************************
*          Stage 8 of 9 - Check for System Services       *
***********************************************************
Attempting to prepare our own version of the bext dependencies

BRL-CAD bext clone command: /usr/bin/git clone https://github.com/BRL-CAD/bext.git
BRL-CAD bext checkout command: /usr/bin/git -c advice.detachedHead=false checkout 06d1ac96f9902f6c931f099b93599e8efcbb06e7
/opt/homebrew/bin/cmake -S;/Users/[MY NAME]/projects/brlcad/build/bext;-B;/Users/[MY NAME]/projects/brlcad/build/bext_build;-DGIT_SHALLOW_CLONE=ON;-DENABLE_ALL=ON;-DUSE_GDAL=ON;-DUSE_QT=OFF;-DUSE_TCL=ON;-DUSE_APPLESEED=OFF;-DCMAKE_BUILD_TYPE=Release;-DCMAKE_INSTALL_PREFIX=/Users/[MY NAME]/projects/brlcad/build/bext_output
/opt/homebrew/bin/cmake --build /Users/[MY NAME]/projects/brlcad/build/bext_build --parallel 8 --config Release

```
The callstack in the error leads to the issue command being `${CMAKE_COMMAND} --build ${BRLCAD_EXT_BUILD_DIR} --parallel ${BRLCAD_EXT_PARALLEL} --config ${CMAKE_BUILD_TYPE}`. This somehow errors. 

I do have an install-out log in the opennurbs/OPENNURBS_BLD-prefix/src/OPENNURBS_BLD-stamp directory and an install-err log that is completely empty. I'm not sure if the error is actually OpenNURBS, but instead whatever comes *after* OpenNURBS because of these logs. Here are the log files and a config log that has CMake warnings for unused options but no explicit errors:

[OPENNURBS_BLD-install-out.log](https://github.com/user-attachments/files/21302141/OPENNURBS_BLD-install-out.log)

[OPENNURBS_BLD-build-out.log](https://github.com/user-attachments/files/21302142/OPENNURBS_BLD-build-out.log)

[OPENNURBS_BLD-configure-out.log](https://github.com/user-attachments/files/21302148/OPENNURBS_BLD-configure-out.log)

[OPENNURBS_BLD-configure-err.log](https://github.com/user-attachments/files/21302165/OPENNURBS_BLD-configure-err.log)

However, if I sort my bext-build folder by creation date, opennurbs is at the top, so I have no idea where the real error is. If I go through the Makefile trail, bext-build/CMakeFiles/Makefile2 says osmesa should be after opennurbs.

I cloned the brlcad build yesterday (July 16, 2025), but I don't remember the exact time or if any new changes have been made since then. Is there a terminal command I can use to find more info about the build version? If that's useful at all. Is there any other log file in the opennurbs folder I should provide instead?

agaztec, 2025-07-18:
Never mind, parallel computing was leading me on a wild goose chase. It's not opennurbs or osmesa. Scrolling up a fair ways through the console I see that *Geogram* is what's failing to build, but it got buried under other modules successfully building. Here are the logs:

[GEOGRAM_BLD-configure-err.log](https://github.com/user-attachments/files/21319053/GEOGRAM_BLD-configure-err.log)

[GEOGRAM_BLD-configure-out.log](https://github.com/user-attachments/files/21319065/GEOGRAM_BLD-configure-out.log)

[GEOGRAM_BLD-build-err.log](https://github.com/user-attachments/files/21319041/GEOGRAM_BLD-build-err.log)

[GEOGRAM_BLD-build-out.log](https://github.com/user-attachments/files/21319049/GEOGRAM_BLD-build-out.log)

Since these issues are resolved on this page (conveniently for Apple Silicon as well) https://github.com/BRL-CAD/brlcad/issues/76 I'll stop posting here.