BRL-CAD/brlcad #188: Compilation error in src/libbg/spsr/SPSR.cpp

thierry-FreeBSD, 2025-05-01:
Trying to compile brlcad on FreeBSD with clang-19 fails with these errors:
```
FAILED: src/libbg/CMakeFiles/libbg-obj.dir/spsr/SPSR.cpp.o 
/usr/bin/c++ -DBG_DLL_EXPORTS -DBRLCADBUILD -DEIGEN_DONT_VECTORIZE -DHAVE_CONFIG_H -I/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/include -I/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/include/brlcad -isystem /usr/local/include/eigen3 -O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing   -isystem /usr/local/include -std=c++17  -D_XOPEN_SOURCE=700 -pipe -fvisibility=hidden -fno-strict-aliasing -fno-common -fexceptions -ftemplate-depth-128 -m64 -g -ggdb3 -Qunused-arguments -O3 -finline-functions -flto -fno-omit-frame-pointer -pedantic -Wall -Wextra -Wundef -Wfloat-equal -Wshadow -Wbad-function-cast -Winline -Wno-long-long -Wno-variadic-macros -Wdocumentation -std=c++17 -fPIC -MD -MT src/libbg/CMakeFiles/libbg-obj.dir/spsr/SPSR.cpp.o -MF src/libbg/CMakeFiles/libbg-obj.dir/spsr/SPSR.cpp.o.d -o src/libbg/CMakeFiles/libbg-obj.dir/spsr/SPSR.cpp.o -c /usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/SPSR.cpp
In file included from /usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/SPSR.cpp:71:
In file included from /usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.h:290:
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:759:35: error: no member named 'center' in 'OctNode<NodeData>'
  759 |                 temp=SquareDistance(children[i].center,p);
      |                                     ~~~~~~~~~~~ ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:819:8: error: reference to non-static member function must be called
  819 |         this->depth = node.depth;
      |         ~~~~~~^~~~~
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:820:33: error: no member named 'offset' in 'OctNode<NodeData>'
  820 |         for(i=0;i<DIMENSION;i++){this->offset[i] = node.offset[i];}
      |                                  ~~~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:829:43: error: reference to non-static member function must be called
  829 |         return ((const OctNode< NodeData >*)v1)->depth-((const OctNode< NodeData >*)v2)->depth;
      |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:836:10: error: no member named 'd' in 'OctNode<NodeData>'
  836 |         if( n1->d!=n2->d ) return int(n1->d)-int(n2->d);
      |             ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:836:17: error: no member named 'd' in 'OctNode<NodeData>'
  836 |         if( n1->d!=n2->d ) return int(n1->d)-int(n2->d);
      |                    ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:836:36: error: no member named 'd' in 'OctNode<NodeData>'
  836 |         if( n1->d!=n2->d ) return int(n1->d)-int(n2->d);
      |                                       ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:836:47: error: no member named 'd' in 'OctNode<NodeData>'
  836 |         if( n1->d!=n2->d ) return int(n1->d)-int(n2->d);
      |                                                  ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:837:15: error: no member named 'off' in 'OctNode<NodeData>'
  837 |         else if( n1->off[0]!=n2->off[0] ) return int(n1->off[0]) - int(n2->off[0]);
      |                  ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:837:27: error: no member named 'off' in 'OctNode<NodeData>'
  837 |         else if( n1->off[0]!=n2->off[0] ) return int(n1->off[0]) - int(n2->off[0]);
      |                              ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:837:51: error: no member named 'off' in 'OctNode<NodeData>'
  837 |         else if( n1->off[0]!=n2->off[0] ) return int(n1->off[0]) - int(n2->off[0]);
      |                                                      ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:837:69: error: no member named 'off' in 'OctNode<NodeData>'
  837 |         else if( n1->off[0]!=n2->off[0] ) return int(n1->off[0]) - int(n2->off[0]);
      |                                                                        ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:838:15: error: no member named 'off' in 'OctNode<NodeData>'
  838 |         else if( n1->off[1]!=n2->off[1] ) return int(n1->off[1]) - int(n2->off[1]);
      |                  ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:838:27: error: no member named 'off' in 'OctNode<NodeData>'
  838 |         else if( n1->off[1]!=n2->off[1] ) return int(n1->off[1]) - int(n2->off[1]);
      |                              ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:838:51: error: no member named 'off' in 'OctNode<NodeData>'
  838 |         else if( n1->off[1]!=n2->off[1] ) return int(n1->off[1]) - int(n2->off[1]);
      |                                                      ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:838:69: error: no member named 'off' in 'OctNode<NodeData>'
  838 |         else if( n1->off[1]!=n2->off[1] ) return int(n1->off[1]) - int(n2->off[1]);
      |                                                                        ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:839:15: error: no member named 'off' in 'OctNode<NodeData>'
  839 |         else if( n1->off[2]!=n2->off[2] ) return int(n1->off[2]) - int(n2->off[2]);
      |                  ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:839:27: error: no member named 'off' in 'OctNode<NodeData>'
  839 |         else if( n1->off[2]!=n2->off[2] ) return int(n1->off[2]) - int(n2->off[2]);
      |                              ~~  ^
/usr/ports/cad/brlcad/work/brlcad-rel-7-40-2/src/libbg/spsr/Octree.inl:839:51: error: no member named 'off' in 'OctNode<NodeData>'
  839 |         else if( n1->off[2]!=n2->off[2] ) return int(n1->off[2]) - int(n2->off[2]);
      |                                                      ~~  ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.

```

starseeker, 2025-05-01:
Urk.  That may be the first attempt to compile with clang-19.  Our SPSR code is an old fork of https://github.com/mkazhdan/PoissonRecon so it probably isn't set up correctly for a newer standard or some such.  

I did make an attempt to update to a newer version - https://github.com/BRL-CAD/brlcad/commit/c36e09e27201af5dcbddd820649b4cb984eb8a99 and specifically https://github.com/BRL-CAD/brlcad/blob/c36e09e27201af5dcbddd820649b4cb984eb8a99/src/libbg/spsr.cpp - but the first cut didn't produce a valid output mesh and I haven't had time to dig into why.

The options would be to figure out why the older code is making the compiler unhappy and figuring out what I did wrong with the newer version to get it to produce correct output instead of the mess I got with that attempt - the latter would be considerably better as a way forward but is also likely harder.

I won't have time to look into this one for a little while - other priorities - but if nobody else is able to run it down I'll circle back to it once I've dug out from my current pile of work.

starseeker, 2025-05-01:
It may be that newer SPSR code has different defaults than the old version, so it might be as simple as figuring out how to replicated the old solver behavior in the newer code with options/compile settings, but that could be a job as well - there are a lot of knobs for that algorithm, not all of them especially intuitive...

starseeker, 2025-07-18:
OK, so good news is Copilot found the issue in my update attempt (stupid indexing typo on my end.)  Bad news is the upstream SPSR API seems to have changed again since I did that experiment - v16.10 works, which is still a big step forward, but to get v18.7 going it'll need some more work.

starseeker, 2025-07-18:
It should be in the next couple weeks, maybe sooner.

starseeker, 2025-09-21:
@thierry-FreeBSD I'm going to go ahead and close this - we have other known compilation issues on BSD in the dependencies, so it may be a bit before I can try the new setup there specifically, but if it's still a problem we can open a new issue.  (Or even submit one upstream, if the problem is in SPSR itself rather than our integration of it.)

closed 2025-09-21 by starseeker

thierry-FreeBSD, 2025-09-21:
OK, thanks for investigating!